MWATelescope/mwalib #10: Should ATTEN_DB be made optional?

cjordan, 2020-09-22:
Not all metafits contain this key. Is this because it was introduced at a particular time? @andreww5au

andreww5au, 2020-09-22:
Yep, I added it in February 2019, because without it, there's no way of knowing what the analogue attenuation was for an observation. In theory, Cotter should be using this key to correctly calibrate an observation taken with one attenuation value (eg a Solar obs) using a calibration observation taken with another attenuation value. In practice, Cotter never added that feature, so the absolute calibration scale is wrong.

cjordan, 2020-09-22:
Thanks @andreww5au. I guess @gsleap and I should work out whether mwalib should treat this key as optional or required, then.  As we (RTS people, cotter people...) have gotten this far without using the key properly, I'm leaning toward optional.  Any software that should use this key could enforce it and prompt the user to get a new metafits file.