MWATelescope/mwalib #22: Change visibility interface to allow access to all timesteps / coarse channels

derwentx, 2021-04-30:
Currently MWALib visibility interface only provides visibilities for timesteps which are common to all coarse channels. This is an issue because sometimes gpubox files start at different times, and the behaviour of cotter is to read these "incomplete" timesteps.

This is an issue, because the gpubox time map could look like this.

| cc \ ts | 100 | 101 |     |
| ------- | --- | --- | --- |
| 0       |     | x   | x   |
| 1       | x   | x   | x   |
| 2       |     |     |     |

There are multiple ways that MWALib currently looks at timesteps in different contexts:
- `MetafitsContext`:
  - `sched_(start|end)_(gps_time_ms|unix_time_ms|utc|mjd)` - scheduled
- `CorrelatorContext`:
  - `gpubox_time_map` - a map whose keys are each unix time from every gpubox, not necessarily contiguous. associates unix time with a map of `gpubox_number` to a tuple of batch number and hdu index.
    - note: do we want to change `gpubox_number` to some other key while we're at it?
  - `timesteps`: a vector of timesteps which are common to the largest number of gpubox files.
- actual start time (correlator context, can start before scheduled start time if common gpubox timesteps start before scheduled start time, which is an edgecase)

There are multiple ways that MWALib currently looks at coarse channels in different contexts:
- metafits file:
  - `CHANNELS` and other headers on fits primary hdu, list of `corr_chan_number`, `rec_chan_number`, `gpubox_number`?,  used by `populate_coarse_channels`?
  - not sure if a gpubox is down, whether metafits file still says it's there.
- `MetafitsContext`:
  - no way to determine coarse channels from metafits context?
  - use `CoarseChannel.get_metafits_coarse_chan_array(metafits_coarse_chans_string)` instead.
- what about when we're missing a gpubox?
- we don't care about handling a gpubox file which is not know about in the metafits.
- need to know if it's legacy or mwax, and voltage or correlator to determine coarse channels from metafits
- leave `MetafitsContext.new` 
- `CoarseChannel::corr_chan_number` - Correlator channel is 0 indexed (0..N-1)
- `CoarseChannel::rec_chan_number` - Receiver channel is 0-255 in the RRI recivers
- `CoarseChannel::gpubox_number` - ambiguous - only makes sense for legacy. in MWAX, this is the rec_chan_number.
- `coarse_chan_index` - index of coarse channel according to `CorrelatorContext.coarse_chans`

We would like to make the following changes:
- `MetafitsContext::timesteps` - all timesteps that the metafits context knows about (from `sched_start` time to ...)
- `MetafitsContext::coarse_chans` - all possible coarse channels (based on metafits headers such as `CHANNELS`).
  - see: `CoarseChannel::populate_coarse_channels`
  - not necessarily contiguous, because of "picket fence" 
  - question: not sure if gpuboxes which are down are included? maybe document this.
- `CorrelatorContext::gpubox_time_map` - should be updated to include data for *all* timesteps that the correlator context knows about. This could be a superset or a subset of metafits, and is not necessarily contiguous.
  - this should be public accessible
  - accessible through FFI?
- `CorrelatorContext::timesteps` - This should be updated to be a union of metafits and gpubox timesteps   
  - this is the "master" timesteps, and is contiguous.
- `CorrelatorContext::common_timesteps` - some interface for the user to determine the timesteps which are common to all provided coarse channels. Must be contiguous, and must all have data.
  - edge case: if a provided coarse channel is empty, then the common timesteps is also empty. 
  - Edge case: if there is a gap in timesteps, everything after that gap is not included in common.
  - Possible implementation:
    - Slice - can this be empty? Yes. - probably the neatest.
    - Vector of TimeStep structs. 
    - Vector of references? - might not be possible.
    - Option<Slice> - Either Some<Slice> or None of empty
- `CorrelatorContext::good_timesteps` - same as common timesteps, but without timesteps before "good time".
- `CorrelatorContext::(start|end)_(unix|gps)_time_ms` - this should now be determined from `CorrelatorContext.timesteps`
- `CorrelatorContext::duration_ms` - this should now be determined from `CorrelatorContext::timesteps`
- `CorrelatorContext::read_by_(frequency|baseline)` needs a new error when there is no data for that coarse channel / time step idx combo.
  - implementation details:
    - new Rust error type, "no data for this combination"
    - FFI will need new return codes.
- `CorrelatorContext::provided_coarse_chans` - just the provided coarse channels:
  - always a subset of `MetafitsContext.coarse_chans`
  - not necessarily contiguous, because of "picket fence" 
  - edge case: do we care if any of these are empty? No, include these anyway.
- all methods which take timestep indices as arguments must be modified so that the timestep indices are based on those from `CorrelatorContext.timesteps`.
- all methods which take coarse channel indices as a parameter need to use the index of the coarse channel according to `MetafitsContext.coarse_chans`.

Nice to have:
- remove any references to the word "actual", this includes comments, and `impl fmt::Display for CorrelatorContext`
- add ability to specify timestep / coarse channel object for reading visibilities
- methods which accept coarse channels could be duplicated to take either `corr_chan_number` or `rec_chan_number` instead of 

Not necessary:
- `(start|end)_good_(unix|gps)_time_ms` - not necessary, can be determine from `good_timesteps.(first|last)`
- `(start|end)_common_(unix|gps)_time_ms` - not necessary, can be determine from `common_timesteps.(first|last)`
- `CorrelatorContext::provided_timesteps` - e.g. if batch 00 and batch 02 were provided, but not batch 01, no known use cases. 
- `CorrelatorContext::new`, specify whether you want:
  - all timesteps
  - common timesteps

Potential issues / edge cases:
- correlator can provide visibilities from before metafits says data should start, e.g. obs_id 1065880128
  - what does cotter do? 
  - negative indices?
  - will anyone ever use this?
  - everything prior to good_time is useless: 
    - receivers are changing in frequency and pointing
  - good_time can never be before metafits start time because minimum quack_time is 0.5s
  - solutions:
    - instead of functions taking a timestep_idx (which could be negative), take a `unix_time_ms` (or `gps_time_ms` or `TimeStep` struct).
      - gpubox time map doesn't care about timestep_idx
- what if a leap second occurs during an observation? do `unix_time_ms` and `gps_time_ms` handle leap seconds the same way?
  - M&C Guarantees that we never do an observation within ~12 hours of a leap second.
- "picket fence" coarse channels
- missing HDUs in middle of gpubox
- missing batches (e.g. provide batch 00 and 02 but not 01)
- provided gpubox with no data
- what happens when a gpubox is down?

derwentx, 2021-06-15:
latest changes in 0.8.1 are working super well for me! Now my outputs match Cotter exactly!

gsleap, 2021-07-01:
Closed by #4d3d40f8b85d99fc28054a030af65b0b132d315c

closed 2021-07-01 by gsleap