MWATelescope/mwalib #26: Change visibility pol to use enums

gsleap, 2021-05-18:
- Change pol to be enum of XX,XY,YX,YY
- Rename string pol to be pol_name 
- Add pol1 to be enum of X,Y
- Add pol2 to be enum of X,Y
- Also add attributes to help Biril convert to MS and uvfits by providing the appropriate order attributes.
So the struct will look like:
{
pol: XX,
pol_name: "XX"
pol1: X
pol2: X
},
{
pol: XY,
pol_name: "XY"
pol1: X
pol2: Y
},
...

cjordan, 2021-05-19:
`mwalib` already has a `Pol` enum. Obviously I'm not at work right now, but I guess Dev wants a `CrossPol` enum? If so, the enum would just have 4 variants. Extra things like the "name" could just be the `.display()` method (see the block at the bottom), and there could be a new method to return two `Pol`s from a `CrossPol`. Just my 2c.

https://github.com/MWATelescope/mwalib/blob/c6b32ef63711f9172111d841da5dd752e3747983/src/rfinput/mod.rs#L77-L106

gsleap, 2021-05-19:
Yep exactly! Basically just adding another enum called VisPol which has the 4 variants and display method :-)

gsleap, 2021-05-20:
Fixed in d63cb0eadf842e866dc72d36ea11cc6669e65fce. Simplified by adding a new enum for VisPol to replace the existing struct.

closed 2021-05-20 by gsleap