MWATelescope/mwalib #64: Update receiver types and implement cable flavour

gsleap, 2023-11-23:
1. Change receivertype enum to have now the following items?

From @andreww5au:
```
Every cable flavour starting with 'RG6', except for 'RG6_90' and 'everything starting with LMR' [has a whitening filter]. 
```

- Unknown,
- RRI_with_whitening_filter -> if rectype=RRI and ((cable_flavour starts with "RG6" AND cable_flavour != "RG6_90") OR cable_flavour starts with "LMR")
- RRI_with_antialiasing -> if rectype=RRI and  NOT ((cable_flavour starts with "RG6" AND cable_flavour != "RG6_90") OR cable_flavour starts with "LMR")
- NI_with_antialiasing -> if rectype=NI
- Pseudo

2. And we want to add another field for each rf_input which is "cable_flavour" which is just the string that AW provides in the metafits?

d3v-null, 2023-11-23:
relevant ticket in Birli https://github.com/MWATelescope/Birli/issues/147

gsleap, 2023-11-24:
Fixed by 22547ed472909ae0fd73ff52b5d99aea6714caad

closed 2023-11-24 by gsleap