MWATelescope/mwalib #67: arm64 error[E0308]: mismatched types

d3v-null, 2024-03-09:
I'm trying to build mwalib for arm64, so that I can demonstrate how to use mwalib in a cross-platform docker container that can run on most typical platforms including Apple Silicon. 
There are arm64 builds provided, but only for macos, which I can't use with a base ubuntu image, so I will need to cross-compile to arm64.

Here's my Dockerfile

```Dockerfile
# syntax=docker/dockerfile:1
FROM ubuntu:20.04 as base

ENV DEBIAN_FRONTEND="noninteractive"
RUN apt-get update && \
    apt-get install -y \
        aoflagger-dev \
        build-essential \
        casacore-data casacore-dev \
        clang \
        cmake \
        curl \
        cython3 \
        fontconfig \
        g++ \
        git \
        ipython3 \
        jq \
        lcov \
        libatlas3-base \
        libblas-dev \
        libboost-date-time-dev \
        libboost-filesystem-dev \
        libboost-program-options-dev \
        libboost-system-dev \
        libboost-test-dev \
        libcfitsio-dev \
        liberfa-dev \
        libexpat1-dev \
        libfftw3-dev \
        libfontconfig-dev \
        libfreetype-dev \
        libgsl-dev \
        libhdf5-dev \
        liblapack-dev \
        libopenmpi-dev \
        libpng-dev \
        libpython3-dev \
        libssl-dev \
        pkg-config \
        procps \
        python3 \
        python3-astropy \
        python3-dev \
        python3-ipykernel \
        python3-ipython \
        python3-matplotlib \
        python3-numpy \
        python3-pandas \
        python3-pip \
        python3-pytest \
        python3-scipy \
        python3-seaborn \
        python3-six \
        python3-sphinx \
        tzdata \
        unzip \
        wget \
        zip \
    && \
    apt-get clean all && \
    rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* && \
    apt-get -y autoremove

# use python3 as the default python
RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 1
RUN update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1

# install python prerequisites
RUN pip install pyvo tabulate matplotlib pyuvdata

ARG RUST_VERSION=1.75

RUN mkdir -m755 /opt/rust /opt/cargo
ENV RUSTUP_HOME=/opt/rust CARGO_HOME=/opt/cargo PATH=/opt/cargo/bin:$PATH

RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y --profile minimal --default-toolchain=${RUST_VERSION}
RUN cargo install --force cargo-binstall

RUN cargo binstall --no-confirm maturin

ARG MWALIB_VERSION=1.2.1

RUN git clone --depth 1 --branch v${MWALIB_VERSION} https://github.com/MWATelescope/mwalib.git /mwalib && \
    cd /mwalib && \
    maturin build --release --features=python && \
    export WHEEL=$(ls -1 target/wheels/*.whl | tail -n 1) && \
    pip install $WHEEL && \
    cargo install --path . && \
    cd / && \
    rm -rf /mwalib
```

And I'm building it with

```bash
docker buildx create --driver=docker-container --name=multi --use
docker buildx build \
  --platform linux/amd64,linux/arm64 \
  .
```

The error message:

```txt
error[E0308]: mismatched types
    --> src/ffi/mod.rs:1246:61
     |
1246 |     let output_slice: &mut [i8] = slice::from_raw_parts_mut(buffer_ptr, buffer_len);
     |                                   ------------------------- ^^^^^^^^^^ expected `*mut i8`, found `*mut u8`
     |                                   |
     |                                   arguments to this function are incorrect
     |
     = note: expected raw pointer `*mut i8`
                found raw pointer `*mut u8`
note: function defined here
    --> /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/raw.rs:141:21

error[E0308]: mismatched types
    --> src/ffi/mod.rs:1381:61
     |
1381 |     let output_slice: &mut [i8] = slice::from_raw_parts_mut(buffer_ptr, buffer_len);
     |                                   ------------------------- ^^^^^^^^^^ expected `*mut i8`, found `*mut u8`
     |                                   |
     |                                   arguments to this function are incorrect
     |
     = note: expected raw pointer `*mut i8`
                found raw pointer `*mut u8`
note: function defined here
    --> /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/slice/raw.rs:141:21
```

d3v-null, 2024-03-09:
this Dockerfile works with mwalib 1.0.1

```Dockerfile
...
ARG MWALIB_VERSION=1.0.1
RUN git clone --depth 1 --branch v${MWALIB_VERSION} https://github.com/MWATelescope/mwalib.git /mwalib && \
    cd /mwalib && \
    maturin build --release --features=python && \
    python -m pip install $(ls -1 target/wheels/*.whl | tail -n 1) && \
    cd / && \
    rm -rf /mwalib ${CARGO_HOME}/registry
```

gsleap, 2024-03-11:
Hi Dev- thanks for reporting this and providing the docker stuff. I'm using that to be able to reproduce the issue- however I think I already know the cause. The function signature that the errors occur in use `char*` - i.e. it not explicitly `signed char`, and [this article](https://stackoverflow.com/questions/20576300/is-char-signed-or-unsigned-by-default-on-ios) seems to indicate that a `char *` in Arm defaults to `unsigned char` (except for iOS which behaves like x86) whereas all other architectures including x86 treat `char*` as `signed char*`. But I will get your docker going first so I can reproduce the error and then  hopefully fix it by removing the ambiguity!

gsleap, 2024-03-11:
Unfortunately I cannot test this on Arm64 as I cannot get your Dockerfile to work on x86 (with some hacks I might be able to get it to work with Arm64). Also since it is a library, `cargo install` does not work. So I'm not sure how your docker file ever worked? Anyway regardless, the issue looks very real so if you can checkout this branch `arm64_fix` and give it another go, I'd appreciate it!

gsleap, 2024-03-12:
Hi @d3v-null, ok this branch has the fix that works. Tested using a heavily modified version of your Dockerfile (I still have no idea how your Dockerfile could ever have worked!)

```docker
# syntax=docker/dockerfile:1
FROM ubuntu:20.04 as base

ENV DEBIAN_FRONTEND="noninteractive"
RUN apt-get update && \
    apt-get install -y \
        aoflagger-dev \
        build-essential \
        casacore-data casacore-dev \
        clang \
        cmake \
        curl \
        cython3 \
        fontconfig \
        g++ \
        git \
        ipython3 \
        jq \
        lcov \
        libatlas3-base \
        libblas-dev \
        libboost-date-time-dev \
        libboost-filesystem-dev \
        libboost-program-options-dev \
        libboost-system-dev \
        libboost-test-dev \
        libcfitsio-dev \
        liberfa-dev \
        libexpat1-dev \
        libfftw3-dev \
        libfontconfig-dev \
        libfreetype-dev \
        libgsl-dev \
        libhdf5-dev \
        liblapack-dev \
        libopenmpi-dev \
        libpng-dev \
        libpython3-dev \
        libssl-dev \
        pkg-config \
        procps \
        python3 \
        python3-astropy \
        python3-dev \
        python3-ipykernel \
        python3-ipython \
        python3-matplotlib \
        python3-numpy \
        python3-pandas \
        python3-pip \
        python3-pytest \
        python3-scipy \
        python3-seaborn \
        python3-six \
        python3-sphinx \
        tzdata \
        unzip \
        wget \
        zip \
    && \
    apt-get clean all && \
    rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/* && \
    apt-get -y autoremove

# use python3 as the default python
RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 1
RUN update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1

# install python prerequisites
RUN pip install pyvo tabulate matplotlib pyuvdata maturin[patchelf]==1.3.2

ARG RUST_VERSION=1.75

RUN mkdir -m755 /opt/rust /opt/cargo
ENV RUSTUP_HOME=/opt/rust CARGO_HOME=/opt/cargo PATH=/opt/cargo/bin:$PATH

RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y --profile minimal --default-toolchain=${RUST_VERSION}

ARG MWALIB_VERSION=arm64_fix

RUN git clone --depth 1 --branch ${MWALIB_VERSION} https://github.com/MWATelescope/mwalib.git /mwalib && \
    cd /mwalib && \
    maturin build --release --features=python && \
    cargo build --release --features examples && \
    cp target/release/libmwalib.a /usr/local/lib/. && \
    cp target/release/libmwalib.so /usr/local/lib/. && \
    cp include/mwalib.h /usr/local/include/. && \
    cd / && \
    rm -rf /mwalib && \
    ldconfig
```

I'll try and do a release in the next day or so :-)

d3v-null, 2024-03-12:
Thanks for the fix Greg. Yes my original Dockerfile had a copy-paste mistake, but the one I posted in the second comment should work right? (just for installing the python library) 

I'm curious why you weren't able to get arm64 emulation working in your Docker, are you using the buildx extension?

Anyway this worked for me.

```
# build with: docker buildx build --platform linux/amd64,linux/arm64 -t d3vnull0/mwa-demo:latest -f Dockerfile .

...
RUN python -m pip install --upgrade pip && \
    python -m pip install \
        maturin[patchelf] \
    ;

...

ARG MWALIB_VERSION=arm64_fix
RUN git clone --depth 1 --branch ${MWALIB_VERSION} https://github.com/MWATelescope/mwalib.git /mwalib && \
    cd /mwalib && \
    maturin build --release --features=python && \
    python -m pip install $(ls -1 target/wheels/*.whl | tail -n 1) && \
    cd / && \
    rm -rf /mwalib ${CARGO_HOME}/registry
```

gsleap, 2024-03-12:
Yeah I think the copy/paste error was the issue with the Dockerfile. I got the arm64 docker working fine (reproducing your error) once I made the mods to the Dockerfile. Your Dockerfile in your recent message fails on this line: `python -m pip install $(ls -1 target/wheels/*.whl | tail -n 1)`. No idea why, but it's superfluous anyway since `maturin build` will already install the Python package to the Python environment.

d3v-null, 2024-03-12:
interesting, I didn't know that it would get installed, and that line works fine in my Dockerfile.

I was thinking for the next release it would be great to provide prebuilt wheels for macos-arm64, since the x86 versions don't work on my mac (unless I'm doing this wrong).

```
> wget https://github.com/MWATelescope/mwalib/releases/download/v1.2.1/mwalib-v1.2.1-macosx-python-x86-64.tar.gz
> tar -zxvf mwalib-v1.2.1-macosx-python-x86-64.tar.gz
> python3 -m pip install mwalib-1.2.1-cp311-cp311-macosx_10_12_x86_64.whl

ERROR: mwalib-1.2.1-cp311-cp311-macosx_10_12_x86_64.whl is not a supported wheel on this platform.
```

To do this, we would need to add a [`macos-14` (beta)](https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners/about-github-hosted-runners#standard-github-hosted-runners-for-public-repositories) job to the release workflow.

d3v-null, 2024-03-12:
Having a matrix of [ py3.8, py3.9, ... ] x [ macos-latest, ubuntu-latest, macos-14 ] might be a good way of avoiding duplicate code. https://docs.github.com/en/actions/using-jobs/using-a-matrix-for-your-jobs

d3v-null, 2024-03-12:
same with `python3 -m pip install mwalib` on my mac.

```
Collecting mwalib
  Using cached mwalib-1.2.1.1.tar.gz (363 kB)
  Installing build dependencies ... error
  error: subprocess-exited-with-error

  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Collecting setuptools
        Using cached setuptools-69.1.1-py3-none-any.whl.metadata (6.2 kB)
      Collecting wheel
        Using cached wheel-0.43.0-py3-none-any.whl.metadata (2.2 kB)
      ERROR: Could not find a version that satisfies the requirement python_version>=3.7 (from versions: 0.0.2)
      ERROR: No matching distribution found for python_version>=3.7
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

gsleap, 2024-03-12:
I think that error is caused by this (below) in my pyproject.toml.

So I have removed the "python_version" (since it is specified elsewhere anyway) and have done a new release (1.2.3) and it is published to pypi. Can you please try again?

[build-system]
requires = ["setuptools", "wheel", "python_version>=3.7"]

Cheers

Greg

d3v-null, 2024-03-13:
Thanks Greg, this works!

```bash
> python3 -m pip install mwalib

Collecting mwalib
  Downloading mwalib-1.2.3.tar.gz (363 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 363.9/363.9 kB 637.1 kB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in /opt/homebrew/lib/python3.11/site-packages (from mwalib) (1.26.3)
Building wheels for collected packages: mwalib
  Building wheel for mwalib (pyproject.toml) ... done
  Created wheel for mwalib: filename=mwalib-1.2.3-cp311-cp311-macosx_11_0_arm64.whl size=851664 sha256=a2bc9d98aa0c2f69e63e58d2618076949ed8a77d1be4ce5bdf1e96663bc0a343
  Stored in directory: /Users/dev/Library/Caches/pip/wheels/ad/11/80/7f627e1f71e02eb63bee02ff7928a7efc882ffd774844deb30
Successfully built mwalib
Installing collected packages: mwalib
Successfully installed mwalib-1.2.3
```

but there is no `cp311-macosx_11_0_arm64.whl` in the releases, so installing from the releases doesn't work yet.

```
> wget https://github.com/MWATelescope/mwalib/releases/download/v1.2.3/mwalib-v1.2.3-macosx-python-x86-64.tar.gz
> tar -zxvf mwalib-v1.2.3-macosx-python-x86-64.tar.gz
> python3 -m pip install mwalib-1.2.3-cp311-cp311-macosx_10_12_x86_64.whl
ERROR: mwalib-1.2.3-cp311-cp311-macosx_10_12_x86_64.whl is not a supported wheel on this platform.
```

Thanks