MWATelescope/mwalib #85: misleading field `corr_raw_scale_factor` for legacy correlator observations

d3v-null, 2024-11-01:
I need to know how much the raw values in the fits files have been scaled by for van vleck corrections.

I had assumed that [corr_raw_scale_factor](https://docs.rs/mwalib/latest/mwalib/struct.MetafitsContext.html#structfield.corr_raw_scale_factor) would give me what I need, as it is referred to in the documentation as 

> Correlator visibility scaling factor used to get the visibilities in Jansky-like units

However, this is misleading. I'll use an example that's included in the Birli repo.

```bash
 fitsheader tests/data/1254670392_avg/1254670392_20191009153257_gpubox01_00.fits | grep -i scale
BSCALE  =                0.125                                                  
BSCALE  =                0.125      
```

the scale factor here is clearly 0.125, but mwalib says it's 1.0

this is because mwalib only looks for RAWSCALE in the metafits, and defaults to 1.0 if it's not found

```rust
        // This is a new metafits key as of Oct 2021. So assume this value is 1.0 unless it is provided
        let corr_raw_scale_factor: f32 =
            get_optional_fits_key!(&mut metafits_fptr, &metafits_hdu, "RAWSCALE")?.unwrap_or(1.0);
```

I need mwalib support for reading the BSCALE values out of the raw files. I believe it's also referred to as SCALEFAC.

[This](https://github.com/RadioAstronomySoftwareGroup/pyuvdata/blob/05ee100af2e4e11c9d291c9eafc937578ef01763/src/pyuvdata/uvdata/mwa_corr_fits.py#L1459) is how pyuvdata does it

andreww5au, 2024-11-01:
Watch out for BSCALE, it's not a normal fits header card, it's been a basic part of the FITS spec since the beginning, and it's usually interpreted along with BZERO at a low level during file io. The numbers you get from the low level read function have probably already had BSCALE/BZERO corrections applied, and they are probably applied automatically for you on write. Those headers exist because fits was integer-only for a long time.

d3v-null, 2024-11-06:
Here's how I've done it in Birli https://github.com/MWATelescope/Birli/blob/6d06754dcec70404e40abe8e4de15e5ce201be5c/src/io/mod.rs#L61

gsleap, 2024-11-06:
A few comments on this:
* `corr_raw_scale_factor` only exists in MWAX observations and was added by Ian in response to a request by Randall to try and get visibilities into "janksky like units". This is nothing to do with BSCALE as such and luckily is mutually exclusive with BSCALE since MWAX does not set BSCALE at all.
* There is a comment in Birli (in the link you sent- line: 92 `according to pyuvdata "correlator did a divide by 4 before october 2014"`) which is worth addressing:
  * The division is not ALWAYS by 4 and it relates to the frequency averaging (or time averaging requested -sorry I can't remember off the top of my head which one it is!). So IIRC the idea is that if you wanted 40 kHz averaging then a BSCALE of 0.25 was applied (assuming it worked on frequency and not time).  For 20 kHz it would be 0.5 For 10 kHz it is 1.0. We'll need to check out a few different pre Oct 2014 obs to verify how this worked to be sure.
  * I think for mwalib- so it is clear what is what, I will leave the `corr_raw_scale_factor` exactly as it is, and introduce a new attribute for the BSCALE and then you can determine which to use and when.

commit dcd82d06 (2024-11-11): Add bscale to CorrelatorContext (for Legacy correlator observations). Fixes #85
closed 2024-11-12 by d3v-null